thejpster/monotron #23: Add drawing routines

thejpster, 2018-09-16:
Add routines for this to the vga-framebuffer crate, and to the API.

- [ ] Line drawing
- [ ] Rectangle drawing
- [ ] Filled rectangle drawing
- [ ] Bitmap copy

thejpster, 2018-10-15:
Unsure if this API should be on the `FrameBuffer` or on some `Canvas` struct which can be plugged into the FrameBuffer (where at the moment we plug in a mutable reference to an array).

MabezDev, 2018-10-16:
If you haven't heard of [`embedded-graphics`](https://github.com/jamwaffles/embedded-graphics) yet, it might be of use. It recently got support for colour displays.

thejpster, 2018-10-17:
I've heard of it, but I don't know how it would tie in with my weird block colour attributes.

MabezDev, 2018-10-19:
Provided you pass a struct that implements the  `PixelColor` trait you can pass whatever you need. Then when you implement the `Drawing` trait, you will get access to all the stuff you need to actually draw that pixel with a colour.

For example in my [ssd1351 driver](https://github.com/MabezDev/ssd1351):

```rust
#[cfg(feature = "graphics")]
impl<DI> Drawing<PixelColorU16> for GraphicsMode<DI> 
    where
    DI: DisplayInterface,
{
    fn draw<T>(&mut self, item_pixels: T)
    where
        T: Iterator<Item = drawable::Pixel<PixelColorU16>>,
    {
        let (width, height) = self.display.get_size().dimensions();
        for drawable::Pixel(UnsignedCoord(x, y), color) in item_pixels {
            if x <= width.into() && y <= height.into() {
                self.set_pixel(x, y, color.into_inner());
            }
        }
    }
}
```

thejpster, 2018-10-19:
Ah, the problem is each colour has to apply to a block of 64 pixels. If you have already used, say, red and yellow within a 64 pixel block, it's not defined what should happen if you then want to draw a blue pixel.