thejpster/monotron #73: idea - use icd/companion chip as second cpu

fanoush, 2019-10-16:
there are two same cpus on the board, both can be reprogrammed via swd so the main one can be i/o video/audio chip with vga frame buffer in ram and the debugging chip can be main cpu as it has almost no i/o except usb console and gpios used as jtag.

The schematics of whole board is in launchpad user guide on TI site.

thejpster, 2019-10-18:
Really interesting idea!

fanoush, 2019-10-18:
Just to let you know that I managed to attach to both cpus via openocd and stlink v2 clone over SWD and managed to backup flash from icdi cpu and erase and restore it to main cpu and when now attached via device usb connector it is detected as icdi interface too  just with different serial number (maybe the real icdi chip takes the serial from main cpu somehow via jtag), here is how it get detected now, first real icdi via debug usb connector then device usb connector
```
[ 8258.972181] usb 1-1.4.2: new full-speed USB device number 16 using xhci_hcd
[ 8259.077559] usb 1-1.4.2: New USB device found, idVendor=1cbe, idProduct=00fd, bcdDevice= 1.00
[ 8259.077573] usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8259.077584] usb 1-1.4.2: Product: In-Circuit Debug Interface
[ 8259.077593] usb 1-1.4.2: Manufacturer: Texas Instruments
[ 8259.077602] usb 1-1.4.2: SerialNumber: 0E1015E1
[ 8259.141823] cdc_acm 1-1.4.2:1.0: ttyACM0: USB ACM device
[ 8259.143261] usbcore: registered new interface driver cdc_acm
[ 8259.143269] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 8276.418142] usb 1-1.4.2: USB disconnect, device number 16
[ 8298.917082] usb 1-1.4.2: new full-speed USB device number 17 using xhci_hcd
[ 8299.022361] usb 1-1.4.2: New USB device found, idVendor=1cbe, idProduct=00fd, bcdDevice= 1.00
[ 8299.022377] usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8299.022387] usb 1-1.4.2: Product: In-Circuit Debug Interface
[ 8299.022397] usb 1-1.4.2: Manufacturer: Texas Instruments
[ 8299.022418] usb 1-1.4.2: SerialNumber: MFG-TEST
[ 8299.031237] cdc_acm 1-1.4.2:1.0: ttyACM0: USB ACM device
```

here is how i run openocd with stlink V2 (I have stellaris launchpad, maybe for tiva C the ID will be different, it that case it will complain about unexpected id)
```
pi@raspberrypi:~ $ sudo openocd -d2  -f interface/stlink.cfg  -c "set WORKAREASIZE 0x8000" -c "set CHIPNAME lm4f120h5qr" -c "set CPUTAPID 0x2ba01477" -f target/stellaris.cfg 
Open On-Chip Debugger 0.10.0+dev-00637-gb3ed97a4 (2019-09-21-21:05)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x8000
lm4f120h5qr
0x2ba01477
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 500 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.248085
Info : lm4f120h5qr.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : accepting 'telnet' connection on tcp/4444

```

for backup and restore I used commands
```
reset halt
flash info 0
flash read_bank 0 stellaris-icdi.bin
```
and
```
reset halt
stellaris mass_erase 0
flash erase_check 0 
flash write_bank 0 stellaris-icdi.bin
```
the swd on the 10pin is reversed when compared to info in pdf, when having board upside down with pins in bottom right corner I see this
```
x   x
 o o
 o o = SWCLK (TCK)
 o o
 o o = SWDIO (TMS)
 o o
  x
```
x=holes, o=pins

Anyway, the main info is that both CPUs are accessible via SWD and are identical and usable so this idea may indeed work:-)

When attaching SWD I  powered the board from 3.3V an GND pins in topleft rows  j1 and j3 and both cpus were powered from that and green led was lit.

fanoush, 2019-10-18:
And BTW  this may be useful too if you have two launchpads
http://processors.wiki.ti.com/index.php/Stellaris_LM4F120_LaunchPad_Debug_How_To
then one can debug one launchpad with another when the ICDI chip is not working. However jtag has too many pins, swd is easier with  that tiny 10pin connector for me so I'd prefer openocd with stlink v2 or even native raspberry gpio pins as swd (which works with openocd  too).